piodabro/WPro58 #3: Compatible with HB5808?

2black0, 2017-08-24:
wow great works, all firmware still working in atmega but this is awesome, stm32 developed

btw, HB5808 (https://hobbyking.com/en_us/hb5808-5-8ghz-40ch-diversity-receiver-for-fatshark-goggles-w-raceband.html?___store=en_us) is using stm32 too, can we flash this firmware into HB5808? 

thanks

piodabro, 2017-09-06:
Have no idea if it work, I've never saw that board. It can be different in "pinology" ;)

2black0, 2017-09-12:
its a board from hobbyking, I see stm32 was used, but yeah IDK wiring of it

piodabro, 2017-09-14:
You can write down the schematic, and then maybe a little changes need to be made to port it to this HB5808.

2hlc, 2017-11-09:
Hi piodabro,

Would be really awesome if ti could be ported over to the HB5808. Regarding the layout I found this: 

https://www.rcgroups.com/forums/showpost.php?p=36930926&postcount=1821

I looked at mine put I couldn't make out what pins are needed for flashing.

piodabro, 2017-11-09:
Someone with this module have to write down the schematic of module.
Grab multimeter and check board traces writing down the connections.

Without that work its impossible to do anything ;)

If you make schematic its probably a little work to make this work on your module (just some pin remapping and setup settings saving on new HW)

2hlc, 2017-11-09:
I have the module so I can get that done. Would High Res Pictures help with traces? If you tell me what to measure on the board I'll do it.

piodabro, 2017-11-09:
Yes You can make Hi-res macro photo of the modules. This could help, but some of connections can be difficult to find on photos.
Just switch your multimeter to buzzer mode (continuity check) and check where goes the MCU pins ;)
Write down connections of every element (draw a electronic schematic).

2hlc, 2017-11-09:
Oki, will do it over the weekend.

yo8aiv, 2017-11-10:
i think he should try to make a dump of the current firmware (i saw the connection pins are available), already. if that works it might be that it will be compatible with this one. 
i don't think these receivers are very different (shure they didn't reinvent the wheel on this one.).

2hlc, 2017-11-10:
I need to desolder the display to get to the main PCB, will do this over the weekend. If I know were the pins are I can make a dump with my ST Linker.

piodabro, 2017-11-10:
If the device has Read Out Protection enabled, then downloading the original FW is not possible.
Disabling ROP on STM32F103 erases the flash, so dont try it ;)

piodabro, 2017-11-14:
If someone send me this module to Poland i can make the schematic on my own and then return module to owner.

2hlc, 2017-11-15:
I didn’t had time this weekend to do it. Hopefully this week I have more spare time. 

/Dre

2hlc, 2017-11-23:
So I got the display off yesterday evening and took a quick picture with my phone. Over the weekend I'll take better pics with my DSLR and trace the pinouts 

[image]

piodabro, 2018-01-05:
Only single side of one board gives nothing.
Closing this issue.

HB5808 version is only possible when someone write down schematics of this boards...

20Sy, 2018-01-11:
Hi, Piodabro I can send you the Rx hb5808.
I have a trouble so I have contact the support hobbyking and they give me a second. 
But there is same problem.
The picture strob (hd3) and lost frame sync when camera send high-light.

piodabro, 2018-01-22:
Thanks to 20Sy.

The HB5808 module is in my hands and now its time to draw the schematic.
At first look the "pinology" of this module is very different than PRO58.
There is no EEPROM onboard, so settings there is need to implement STM32 flash EEPROM emulation.
Lots of resources could be easily remapped to work with HB5808.

Stay tuned ;)

piodabro, 2018-01-22:
The bad news is that module doesn't have FatShark button support (by hardware design).

piodabro, 2018-02-02:
Working on code now, there is some thing to #ifndef because some functionality is not posible in HB5808.
Got compiling untested code now. Its time to test on 20Sy's module ;)

All schematics and code will be commited with working HB5808 version.

piodabro, 2018-02-07:
It's almost done ;)
I had to implement FLASH EEPROM emulation to store user settings.
It works, now time for tests ;)
[image]

2hlc, 2018-02-07:
Thanks a million for the work you are doing. How is the module on the hardware side compared to other models like the LaForge? Is the Hw good and only let down by the software the used?

/Dre

piodabro, 2018-02-07:
I have't tested this in-flight so i cant tell is it good or bad.
Compared to PRO58 this module has lack of few features like buzzer, FS button support, EEPROM(this one will be emulated in STM32's internal flash).
But it has USB port so maybe it could be used for some new features like configurator or firmware upgrade.

On my desk it seems to work ok with new firmware but it needs a lot of testing.

The same as PRO58, if you flash new firmware, there is no way back to restore original.

20Sy, 2018-02-07:
Very Good working

piodabro, 2018-02-08:
And its done!
Go [here](https://github.com/piodabro/WPro58/releases), download HB5808.bin, flash and enjoy.

Flashing HB5808 procedure is simply the same as PRO58. SWD connector is decribed [here](https://github.com/piodabro/WPro58#quanum-hb5808-pinouts)

closed 2018-02-08 by piodabro

20Sy, 2018-02-08:
I try it this afternoon. Many thanks Pio