piodabro/WPro58 #4: Feature request : how to edit - LOGO"/Callsign in screensaver mode !

yo8aiv, 2017-09-24:
Yep that's exactly what i meant on rcgroups.
Feature: **how to edit - LOGO"/Callsign in screensaver mode** !

Very nice job.
Cheers.

piodabro, 2017-11-01:
It will be possible when I release open-source code. Then you have to compile your own version with changed logo in header file.

piodabro, 2017-11-09:
In the tools dir is a tool to create byte array from graphics (B&W BMP file, 128x64pix).
Just make your own logo and use the tool to convert it to code.
In this tool set byte order to horizontal.

Then change the logo.h header file and now you have your own logo ;)

yo8aiv, 2017-11-17:
Great. Thanks for the instructions.
Maibe somebody with some C++ skillz can make a utility app for that purpose.
Dziękuję.

mihero, 2017-11-19:
Hi, I have first version done for eeprom stored callsign in sceensaver.
New settings screen for changing it.
https://github.com/mihero/rx5808-pro-diversity-pro58/tree/feature/callsign
Screensaver screen needs still bit tuning.
Any proposal for additional info into callsign screen?
I was thinking to add current diversity mode.
callsign is limited to 8 char.

piodabro, 2017-11-29:
If you think its ready to publish, then make pull request ;)
Once i test it, we can make new version (FS buttons and callsign)

mihero, 2017-11-29:
This has been a bit on hold.
I will finish it with just editable callsign.
Add other stuff then on another pr if needed.

piodabro, 2017-12-05:
Editable callsign done.
Big thanks to mihero!

Closing

closed 2017-12-05 by piodabro