piodabro/WPro58 #8: Feature proposal: interupt based diversity calculation

mihero, 2017-12-02:
To ensure rssi calculation is not affected by other functionalities it could be handled as a separate interupt. This would give control of frequency of diversity calculation.
Also dma for rssi could be changed to be compile time controllable so that dma would store more then 1 values and then calculate average of those to be used on diversity calculation.  This would clean rssi values.

piodabro, 2017-12-02:
For diversity calculation we could use one of timers interrupt to make it time constant.
But first it could be good to make main loop timing measurement. If it loops faster than diversity calculation delay it wont be necessary ;)

The RSSI values are constantly updated via DMA. Of course this should be improved using a little filtering that will give much better results, i.e. Yi = a * Xi + (1- a)*Yi-1 where 0 < a <, Xi is current ADC value, Yi is current filtered value, Yi-1 is last filtered value.
It will give some RSSI reading smoothing and improve diversity switching.

piodabro, 2017-12-03:
I've measured looptime and it is <= 1ms.
I think its not necessary to move diversity switching to interrupts.
but some other things can be improved like I2C on DMA.
I will make some changes in RSSI reading.
I will use TIM3 to make 400Hz interrupts to read RSSI values from memory (written from DMA), and apply some filtering.

mihero, 2017-12-04:
Ok, sounds fast enough loop.
Filter wise should it be 2 phase, like 1 dimensional kalman filter for both rssi reads independently and then exponential filter for rssi diff calculation. This would provide clean rssi values and then control how quickly diversity switch happens.
I already started creating additional state view for receiver "status" monitoring, that could contain measurement errors, current rssi values and then the current filter constants.
Although having some issues on printing rssiA/B values from the receiver.

piodabro, 2017-12-04:
I think it's too complicated filtering.
My intention to use simple averaging filter is removing accidental ADC errors from RSSI signals.

Making RSSI readouts with perdiodically called timer interrupts can give us also place for constant time diversity switching code calls that You proposed.

mihero, 2017-12-04:
sounds good.  
I agree that better to start with simpler working solution then with overly complex.

mihero, 2017-12-04:
Got the receiver status page working.
https://github.com/mihero/rx5808-pro-diversity-pro58/tree/feature/receiver_status
Seems that dma writing completes somewhere 50-80 times before every rssi update.

philbertphotos, 2018-01-06:
> is this firmware finaly the one that let this pro58 work how it shut be??
> i have this thing laying for nothing because its NOT working... so help?

It works fine i have 2 pro58 with version v5 and working without issue.

piodabro, 2018-01-08:
Current version works ok, this issue is to make it working better.
Current solution wroks, and if your module is free of hardware problems it will work just fine.