piodabro/WPro58 #11: Project structure rebuild

piodabro, 2017-12-05:
I've done project structure rebuild including new HAL peripherial configuration.
@mihero  once you confirm it works for You i will merge it to master branch

https://github.com/piodabro/rx5808-pro-diversity-pro58/tree/fix/project_structure

mihero, 2017-12-05:
Looks good.
Compiles fine and works in fatshark dom v3.

- Some not needed indent changes in main.c
- gitignore missing
- new dma channel and tim4 but i assume that those are meant for rssi improvements coming later.

piodabro, 2017-12-05:
Theres should not be main.c ;) only main.cpp

TIM4 was already used for buzzer wave generation (buzzer is passive).
TIM3 is the new timer that generates interrupt at 400Hz. (I think it could be faster, for ex 1kHz).

New DMA channel is for display I2C RX (i dont know if its necessary).

Merging to master

mihero, 2017-12-06:
I'm new with stm32 board, could you help understanding how the new added timer3 is connected to rssiA/B adc and dma. Tried to read the code but not fully understanding how those are connected.

piodabro, 2017-12-06:
There is no connection now. 
It will be just interrupt that will pickup values from rssiDMARaw array, filter and write to rssiARaw/rssiBRaw variables.
DMA will circulary write to rssiDMARaw array from ADC.
Only trick is to synchronise them. Stop ADC reading when we read the array in interrupt and enable right after.
Also i've increased ADC conversion time to 239,5 cycles.

Interrupt is not implemented yet.

mihero, 2017-12-06:
ok, that explains it.
thanks

closed 2018-02-08 by piodabro