piodabro/WPro58 #15: OLED flip screen feature

mgrapiniak, 2018-01-04:
Please implement flip screen feature.
Automatically or manually set.

mihero, 2018-01-04:
pro58 doesn't have any orientation sensors so only possibility is manual.
picked from forum that request is to have 180degree orientation change but feature could include also 90 and 270 degree orientation change

piodabro, 2018-01-05:
Yep, but rotating vertically needs to reimplement views to vertical 128 x 64 screen.
I think we should start with simple upside-down rotating, and then edit status clasess to support vertical views finally adding 90/270 deg options.

mihero, 2018-01-05:
True. simple upside down should be easy task.
supporting 90degree would most likely need redesign on graph screens.

mgrapiniak, 2018-02-24:
Hey guys,
Any chance to implement this feature in upcoming release? How would you estimate work needed for making this happen? Perhaps I could submit PR but to be honest I would require some help :)

piodabro, 2018-03-01:
This have to wait until i make scrolling settings menu (cause it can handle only 8 entries on single screen)

closed 2018-03-04 by piodabro