piodabro/WPro58 #19: feature request

stevefoxuk, 2018-02-07:
A setting in the menu system to prevent the alternating callsign/channel screen from comming up to allow for the graph screen to be permanently displayed.
[image]

piodabro, 2018-02-10:
Feature will be part of next release. Its now done in current code, but needs some tests.

stevefoxuk, 2018-02-10:
yay, thankyou!  when you say tests, do you mean to check stability? im willing to test pre release in a home enviroment if you want, i can leave it running for periods of time, with and without a vtx running.
i cant test in real world at the moment as the weather here is about as bad as it gets right now lol

i would also be prepared to purchase another module just for running test firmware is that would help in the future?

stevefoxuk, 2018-02-10:
scrap the "willing to purchase" part, ive just bought another as a spare from banggood so in a couple of weeks will have a 3rd module ha ha

philbertphotos, 2018-02-11:
I feel like buying another one myself :)

stevefoxuk, 2018-02-11:
yeah, at the price they are its a no brainer!
they are sold as fatshark modules but it dont matter what you need a reciver for, they spank everything!

ive now got a bunch of what i class as 'junk' receivers that cost the same or more than these.
Two are just bog standard led display receivers and  one is a skyzone true diversity but even thats a pile of crap compared to these pro58's and the skyzone gets hot enough to cook on if you dont regulate the voltage.

It supposedly can take 7.4 to 24v, yeah right! at just 12v it was burning hot and has discoloured the pcb! 
the funny thing is that its the internal voltage regulator that was getting dangerously hot, i think it would actually catch fire if you tried putting 24v through it, so i regulated it to 8v as thats a happy voltage for my monitor too and it was convinient to run both from the same line.

The pro58 turned out to be convenient too as i also have a 5v regulated source for my dvr on my groundstation.

i use a monitor as i cant focus when using goggles, ive got box goggles because of the fov but even after modifying the frenzel lense adustment, i still cant comfortably focus, im gonna be booking an eye test as ive noticed im not able to focus on lots of near objects recently, all started after i had a bunch of massive headaches, think i need to get checked out lol.

piodabro, 2018-02-11:
If you can't wait for release, you can build current version from source ;)
I plan a little more work to do before release new version.

stevefoxuk, 2018-02-12:
i wish i could, im not as smart as you, well maybe i am but just not in this field!

closed 2018-02-23 by piodabro

stevefoxuk, 2018-02-23:
big thanks to implementing this.

steve