piodabro/WPro58 #22: option to make fatshark button pins replicate jog wheel

stevefoxuk, 2018-02-18:
an option to make the fatshark button pins activate and navigate the menu system would be usefull.

1 pin for select and the two other pins for up and down/ left right in the menu system.
allowing external buttons to replace the fragile and fiddly jog wheel.

piodabro, 2018-03-01:
To add this feature in settings menu i have to work on settings menu scrolling (cause screen can handle only 8 entires). 
This can be done fast as a separate .bin file with changed buttons pins to FS pins (compiled with no possibility to change in menu).

stevefoxuk, 2018-03-03:
So are you saying it can be done in a seperate release without menu change.
Or...
It can be done with the new menu but your not going too.
Or.....
It can be done with the new menu and you will but it will take time?

I'm a little confused ha ha

Steve.

piodabro, 2018-03-04:
Menu scrolling is almost done ;)

closed 2018-03-04 by piodabro