piodabro/WPro58 #23: Feature - OSD on screen show RX rssi levels !

yo8aiv, 2018-02-22:
Can it be done !! ?
OSD on screen (inside googles) show rssi levels  for A + B .. this way we can see while flying how anntennas switch based on proximity and antenas.

others have done it .. so i'm keen to try do the hardware mod .

Thanks for the Great work until now.

piodabro, 2018-02-23:
> others have done it ..

who?
In PRO58 or HB5808 there is no hardware to do this.
Achilles done "OSD" with a little dirty hack, but this is not On Screen Display, this is override of video signal that causes blank screen with white text ;)

I've got an idea how to make real OSD screen bar with simple info.
But this needs a little bit more modifications - We need to capture composite video sync signal.

Now i'm waiting for some chips that can separate sync signals form video line.
But form china it will take some time ;)

stevefoxuk, 2018-02-23:
whilst in the 'idea' stage, does any of it pose a risk to the unit itself or is it more just a case of separating and merging the signals for the video and what Achilles does in the separate screen display

i know i make it sound simple and theres a lot of knowledge in terms of code and electronics to be able to actually select the right components and make it actually work but what i mean is electrically, my understanding of the outputs are they are reasonably resilient and  this will ether work or wont work but no 'frying' risk?

if you wanna tell me the components needed i would order them regardless of weather you know if they will work yet, then youl have someone ready to try and test the mod on another unit instead of the release comming out then a big wait for people to get ic's from china :)

i dont care if it comes to nothing, ic's never come to more than a couple of quid a piece  and if it dont work they will just end up in the 'hmmm, that might come in handy one day' box ha ha

yo8aiv, 2018-02-23:
well if this can be done, i'm shure it will be possible more than just the simple bar screen .. wich for me is more than enough.
**i'm in for testing if needed. just let me know what to order.** 
i think i saw this sort of osd on some other new module not achillez fw .. 
im just worried on power draw. at this stage module is taking 380mAh to 400mAh

piodabro, 2018-03-10:
I'm waiting for components from china. I will publish a component list and modification information when I check if it works properly. (Once I can generate any overlay on screen).

stevefoxuk, 2018-03-10:
aww, cant we know the list in advance and take a chance that the components may end up being useless if you dont succeed?

im watching this very closely as i have new goggles coming (been using a monitor till now as my first goggles sucked lol)  and ill be modding them for the pro58, so osd would be even more useful to me then!

piodabro, 2018-03-12:
I want to use a popular composite video sync signals separator chip (LM1881), but first i want to know if this mod can fit somwhere over the PRO58 module. This will be a little more components than 2 resistor which you can hide in module.

stevefoxuk, 2018-03-12:
i can get the LM1881 in both regular and smd from the uk for very little.

when you say a little more components than 2 resistors, how many components are you talking?

the smd LM1881 is tiny :)

stevefoxuk, 2018-03-12:
depending on how many other components are required, theres enough space on the back of the module between the antenna solder points for a smd 1881 and a little more :)

stevefoxuk, 2018-03-13:
can i have a shopping list for the other components required?

stevefoxuk, 2018-03-22:
Hey Pio, you get your 1881 yet?
Im not impatient for an osd, i just want to make something ha ha

tun0, 2019-06-17:
Did this ever go anywhere?

piodabro, 2019-06-17:
@tun0, three is a branch OSD with working OSD and LM1881 additional module, but it needs more work to release stable version.

tun0, 2019-06-17:
Ah, nice. Hardware is looking pretty doable. I assume the work to be done is on the software side of things? Or are changes expected on the hardware side as well?
Also, I guess it isn't under active development right now, given that the last commit was 8 months ago?

piodabro, 2019-06-17:
I'd rather say that its a sleepy development right now. IMO The OSD code needs some cleanup to be in release quality. I have not so much time to work on this kind of projects right now.

stevefoxuk, 2019-06-17:
Hey pio,  if anyone knows if this can be done, it's you.

Is it possible to to cut traces from the stm ic and solder to the 12c pins to drive an external oled screen?

I'd like to use one of my modules on an antenna tracker and have an oled screen for the pro58 mounted on the trackers electronics box next to the oled for the tracker so that both displays are easy to see  and I'll mount control buttons on the box too.

The stm ic has 3 separate 12c outputs from what I can see and I'm thinking if I figure out what output the screen is using and isolate it,  I can re route it to a different 12c oled?

Thanks, Steve.

piodabro, 2019-06-19:
@stevefoxuk Yes, it can be done, but the best way to do this is desoldering original display and use its i2c pin outputs.

stevefoxuk, 2019-06-19:
Do you mean soldering to all 30 pins the actual screen uses or do you know the points where it's the two SDA and SCL lines?

Steve.

piodabro, 2019-06-19:
I mean just grab module with i2c interface like this:

[image]

and wire up power (VCC, GND) to proper 3.3V and GND on RX.
Then wire CLK and SDA to pads 11 and 12 of LCD ribbon.

yo8aiv, 2019-07-28:
after buying the ev200d googles and seeing its superior reception of those rx's .. pro58 kind of stay in a box .. but i might also  use it for a ground station or a tracker soon.  
still has cool features.  so i'm waiting for some testing.