piodabro/WPro58 #30: 3D mode

Mrkvozrout, 2018-03-18:
Hi, does this FW support 3D mode? Maybe I just missed it but I did not find it.

My FatShark goggles can be switched to 3D mode, but until now I had only stock VRX module with one antenna. Now I bought Eachine Pro58 module and I would like to try to fly in 3D with two omnidirectional antennas.

philbertphotos, 2018-03-18:
3D mode requires two FPV cameras

Mrkvozrout, 2018-03-18:
Right, I have several cameras so now I am interested about the receiver part.

stevefoxuk, 2018-03-20:
any receiver can do 3d, you just need two of them, all 3d systems have two receivers two cameras, two transmitters and compatible goggles.

stevefoxuk, 2018-03-20:
have you come here from the rc groups page?

if not, come talk to us there, you will get faster replys- https://www.rcgroups.com/forums/showthread.php?3030923-WPro58-diversity-firmware

Mrkvozrout, 2018-03-20:
I have Fatshark Attitude V3 goggles which have just one module bay, but according to the manual they support 3D mode. There is a HW switch on top that switches 2D and 3D mode.

They say "interlaced 3D support" so I quess odd lines for left eye, even eyes for right eye or vice versa.

So Pro58 HW cannot handle video from both antennas simultaneously (just measuring signal strength)? Or is it too demanding for the CPU?

(thanks for rcgroups link)

stevefoxuk, 2018-03-20:
progressive scan and interlaced, ie 576p and 575i (PAL) or 486p and 486i (NTSC) is the way a display draws an image to the screen with interlaced the display shows even lines of 1/60th of a second then the off lines for 1/60th of a second per frame.

if somehow the googles took odd and even lines and split them to each eye, you would be literally halving your screen resolution!

but this is not how its done.
you need 2 cameras connected to 2 transmitters on different frequency and 2 receivers on those 2 frequency's.

the pro58 has one video output, as do all the other modules i know of, they cannot output 2 video sources even if they wanted too, its not to do with cpu load and infact the PRO58 has the most powerful cpu of all these modules :)

ive looked at the user manual for your goggles and it dont help much does it ha ha nor does google searching the subject.

im as confused as you are now, the only thing i can see is that the manual states switching between analogue mode and 'something else' when in 2d or 3d modes.

i will try and find out for you but try to come to rc groups and discuss the problem with us on the WPRO58 page there, its easier and will be seen by more people too!

https://www.rcgroups.com/forums/showthread.php?3030923-WPro58-diversity-firmware

stevefoxuk, 2018-03-20:
this link may be of help to you, seems the goggles do use line seperation for the 3d, this must half the quality!

https://www.rcgroups.com/forums/showthread.php?2540555-3D-FPV-cameras-for-video-goggles-Fat-Shark

piodabro, 2018-03-21:
PRO58 Hardware doesnt allowo to split video signals.
Both receiver outputs goes through multiplexer switched by MCU.
There is no way to separate signals without hardware modification.
If your goggles separates odd/even lines then there is nothing to do for PRO58. It just receive camera signal and pass it to goggles (with switching to better received signal strength).

closed 2018-03-21 by piodabro

Mrkvozrout, 2018-03-21:
OK, thanks for information.