piodabro/WPro58 #31: Fine tune freq on disabled Rx

pmarkiewicz, 2018-03-20:
Channels have hardcoded frequencies, but I don't belive that Tx don't drift and all time is spot on channel freq.
So idea is on disabled recv shift freq in small steps in small range and remember freq with highest rssi.
Then use new found freq after rx switch.

piodabro, 2018-03-21:
It is possible with PRO58 modules, but impossible with HB5808 modules. HB5808 has single CS line for both receivers. So we cant talk to only one RX5808 module :(
But this is quite good idea. I'll give it a try, but low priority...

pmarkiewicz, 2018-03-21:
if you don't mind I can start looking into that, but I will probably need
some help :)

regards

stevefoxuk, 2018-03-21:
thats what one of the other FW's does, its a good idea and seems to work well, just dont like that FW as a whole so use good old WPRO :)

RishiThakur, 2018-07-05:
Is this the request to allow for altering frequencies in 1mhz increments?