piodabro/WPro58 #33: README.md doesn't list any features or explain what this firmware does

ahawks, 2018-06-28:
I saw WPro58 listed as a firmware option for the Eachine Pro58, but I'm having a really hard time finding anything that actually says what the firmware does compared to the stock firmware, or why I'd want to use it.

piodabro, 2018-06-29:
Compared to stock? It works  ;)
Stock has problems with diversity algorithm.
This firmware is really simple in use, and flashing it doesnt prevent You to try other 3rd party firmwares with this module (such as Pirx or Achilles Plus).
Of course i will prepare a features list...  ...other day ;)

stevefoxuk, 2018-07-12:
The stock firmware is broken, the diversity does not work as it should, WPRO58 fixes it and diversity works.

WPRO also has a vastly improved user interface along with extra features not available in the stock firmware like screen flip and the ability to use buttons instead of the jog wheel along with other things.

in every aspect, WPRO is vastly better than the stock firmware.

trust me, you wont regret flashing WPRO, it spanks the stock!

ahawks, 2018-07-12:
I've already flashed WRO58. But I do think what @stevefoxuk just said would suffice to put in the README.md, for new users, or people trying to decide what firmware to use.

philbertphotos, 2018-07-12:
FYI the newest stock firmware (v2 i guess) does much better diversity... but WPRO is still wayyyyyyyy better