piodabro/WPro58 #37: Power up in favorites when last mode was favorites

pitts-mo, 2018-10-08:
I am requesting this feature after finding this desirable when I attended an event utilizing with a channel set across multiple vendor channel sets.

piodabro, 2018-10-15:
I think that the best solution is to save this setting to eeprom when someone enter Search or Fav screen and default with that on startup. Entering other screens won't affect "startup screen" setting.

pitts-mo, 2019-09-13:
I finally had a few minutes to look into this (my local yearly event is coming up). As a temporary fix I have commented PRO58/Src/main.cpp line 185 and added a new line below it as shown.

//	StateMachine::switchState(StateMachine::State::SEARCH);
	StateMachine::switchState(StateMachine::State::FAVOURITES);

When more time allows I hope to poke at:
-Add allocation of eeprom space for variable type bit or ???
-Add distinct updates to this new variable on entry to SEARCH and FAVOURITES.
-Add checking into the above section of PRO58/Src/main.cpp to auto load state defined by new variable value.
-Add fail-safe to pick a state if this eeprom value does not match expected value (corrupted).