piodabro/WPro58 #41: Please document recommended build environment

pitts-mo, 2018-12-20:
I believe WPro58 would benefit from build environment documentation and/or a docker-ized build environment if someone has time/ability.

I may be going about this the wrong way... But I have recently discovered that docker-ized build environments available for betaflight and opentx have drastically reduced my barrier to entry and allowed me to make and test simple code changes on my own.

piodabro, 2018-12-20:
All thing needed to build it is System Workbench for STM32 (SW4STM32).
This is Eclipse based IDE.

pitts-mo, 2018-12-22:
As soon as I read this I remembered it sounded familiar.
This is already documented in README.md 
"To edit/build code you need to download SW4STM32 IDE (Eclipse based)"
Thank you and Sorry, -p

closed 2018-12-22 by pitts-mo